MicrosoftDocs/windows-driver-docs #2227: Return Value 259 is missing

CharlesNRU, 2020-05-05:
I just used PNPutil to install drivers on a machine and the exit code returned was 259.
Could we get a definition of what exit code 259 means for PNPutil?
Thank you.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3f814824-8b1b-766f-8fbb-86a16996eff6
* Version Independent ID: 144f5337-ec84-3587-8caf-cfae4635dc42
* Content: [PnPUtil Return Values - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/pnputil-return-values#feedback)
* Content Source: [windows-driver-docs-pr/devtest/pnputil-return-values.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/pnputil-return-values.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

rodwiddowson, 2020-08-18:
FWIW 259 == 0x103 == STATUS_PENDING.   May not be relevant

tedhudek, 2020-10-23:
Sorry for the delay, I haven't been able to find any more info for you about error code 259.  If you click through from the page you referenced to https://docs.microsoft.com/en-us/windows/win32/debug/system-error-codes--0-499-, that page says 

```
ERROR_NO_MORE_ITEMS
259 (0x103)
No more data is available.
```

If it's still an issue, you could try asking on the forums:

* [Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
* [OSR's ntdev List](http://www.osronline.com/showlists.cfm?list=ntdev)

diogocatossi, 2020-12-18:
I've cam through this issue today and it happened on the second update request for the same driver (while testing a deployment script) so I can infere that's related to the driver being already being updated and no action is required.

peterchen-cp, 2021-07-01:
Some testing suggests: 
259 = "ERROR_NO_MORE_ITEMS" is returned, if the driver was updated on at least one device.
(this is the return code when enumerating devices, and the end of the list is reached.)

So this seems like a bug in pnputil, and 259 should be treated as "success".

tedhudek, 2021-08-03:
Hey folks - I went ahead and added return value 259 to the doc page.  Please take a look at https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/pnputil-return-values.  Thanks again for bringing this to our attention.

closed 2021-08-03 by tedhudek