MicrosoftDocs/windows-driver-docs #2228: Add hex value to table

sarman1998, 2020-05-05:
It would be very useful if this table included the hex value of each minor function.   
Either in a separate column or just after the name.
Ex.  IRP_MN_STOP_DEVICE(0x04)
Also would be good to include that on the page for each one.
If you are concerned about the values changing in the future, then at bare minimum, a link to the header file that contains the definitions so it could be accessed quickly.

Thanks,
  Eric

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 40e6f236-d5a2-aab5-7a6d-c0be6282e5a6
* Version Independent ID: c32901df-1a3f-3ed1-1bad-250f0dbf4b17
* Content: [Plug and Play Minor IRPs - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/plug-and-play-minor-irps#feedback)
* Content Source: [windows-driver-docs-pr/kernel/plug-and-play-minor-irps.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/plug-and-play-minor-irps.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-05-06:
Thanks @sarman1998 that's a good suggestion. The hex values are in wdm.h, in the WDK. I will add them to the docs. Curious, what's the scenario where they help?

Here they are, for your reference until I can make the doc change, from the 19H1 wdm.h:

```
#define IRP_MN_START_DEVICE                 0x00
#define IRP_MN_QUERY_REMOVE_DEVICE          0x01
#define IRP_MN_REMOVE_DEVICE                0x02
#define IRP_MN_CANCEL_REMOVE_DEVICE         0x03
#define IRP_MN_STOP_DEVICE                  0x04
#define IRP_MN_QUERY_STOP_DEVICE            0x05
#define IRP_MN_CANCEL_STOP_DEVICE           0x06

#define IRP_MN_QUERY_DEVICE_RELATIONS       0x07
#define IRP_MN_QUERY_INTERFACE              0x08
#define IRP_MN_QUERY_CAPABILITIES           0x09
#define IRP_MN_QUERY_RESOURCES              0x0A
#define IRP_MN_QUERY_RESOURCE_REQUIREMENTS  0x0B
#define IRP_MN_QUERY_DEVICE_TEXT            0x0C
#define IRP_MN_FILTER_RESOURCE_REQUIREMENTS 0x0D

#define IRP_MN_READ_CONFIG                  0x0F
#define IRP_MN_WRITE_CONFIG                 0x10
#define IRP_MN_EJECT                        0x11
#define IRP_MN_SET_LOCK                     0x12
#define IRP_MN_QUERY_ID                     0x13
#define IRP_MN_QUERY_PNP_DEVICE_STATE       0x14
#define IRP_MN_QUERY_BUS_INFORMATION        0x15
#define IRP_MN_DEVICE_USAGE_NOTIFICATION    0x16
#define IRP_MN_SURPRISE_REMOVAL             0x17

#if (NTDDI_VERSION >= NTDDI_WIN7)
#define IRP_MN_DEVICE_ENUMERATED            0x19
#endif
```

sarman1998, 2020-05-07:
Hi Ted,
   I was debugging using some captured logs from the windows inbox USB driver stack and decoding irps. I found this doc page but couldn’t see the values.  Had to open up Visual studio and find them that way.  

Thanks!
  Eric

PS bring back Microsoft Message Analyzer!

tedhudek, 2020-05-07:
Hey Eric, I just pushed a commit with the changes you requested. Should be live in a couple minutes. Do you think there's value in doing the same thing for pwr mgmt and WMI minor IRPs?

sarman1998, 2020-05-07:
Hi Ted,
   I think it could be useful but where do you stop? :)  does MSFT publish wdm.h on github?  If so just a line somewhere in the article that says 
“Defined in wdm.h”. With wdm.h as a clickable link to the actual file on github would be sufficient. That way any reader can quickly find the actual values if they need. 

 Best Regards,
  Eric

tedhudek, 2020-07-22:
Closing this one out since the work is done.

closed 2020-07-22 by tedhudek