MicrosoftDocs/windows-driver-docs #2233: Firmware update USB device sample - link is broken

tewarid, 2020-05-06:
"Firmware update USB device sample" link doesn't lead to a valid sample app. Through some searching found the following archived sample app but I am not sure if it is still valid. I am looking for a working sample for DeviceServicingTrigger. Any help is appreciated.

https://github.com/microsoftarchive/msdn-code-gallery-microsoft/tree/master/Official%20Windows%20Platform%20Sample/Windows%208.1%20Store%20app%20samples/%5BC%23%5D-Windows%208.1%20Store%20app%20samples/Firmware%20Update%20USB%20Device%20sample/C%23.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e46710c3-f9c4-279f-7134-2ef0610ea028
* Version Independent ID: 7210cc0b-ef33-0beb-4e0c-7a4490b82038
* Content: [Device sync and update for Store device apps in Windows 8.1 - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devapps/device-sync-and-update-for-uwp-device-apps?redirectedfrom=MSDN&source=docs)
* Content Source: [windows-driver-docs-pr/devapps/device-sync-and-update-for-uwp-device-apps.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devapps/device-sync-and-update-for-uwp-device-apps.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

RAJU2529, 2020-05-07:
@tewarid . I conformed that the link is broken . i could not able to find an alternative link

EliotSeattle, 2020-05-07:
Thanks, @tewarid for bringing this to my attention.  
This older sample is, indeed, no longer in the catalogue. I am researching if it was removed purposefully or if this was an error.
Eliot

tedhudek, 2020-10-22:
Hi @tewarid thanks for reporting this, and apologies for the delay. I checked around and was unable to find the sample in question, so I have removed those links.

closed 2020-10-22 by tedhudek