MicrosoftDocs/windows-driver-docs #2234: FAILURE_BUCKET_ID:  0x139_1e_INVALID_NEXT_THREAD_nt!KiFastFailDispatch

HollisTech, 2020-05-06:
Please document what parameter 1 1E means for BUGCHECK 0x139. If it is actually that the next thread is invalid, a hint regarding what the invalid thread object address is would be helpful. At least with that data debug logs might be useful for establishing a root cause.

RAJU2529, 2020-05-07:
@Cheoljin . Which article are you referring?

DOMARS, 2020-12-08:
@HollisTech - Thanks for this request. I don't see a value of param 1 equal to 1E, so will close this issue. Let me know if this is something you have seen in !analyze -v. You can see the known values for 0x139 using !analyze -show. You may be aware that the thread commands in the debugger can be used to look at the thread objects. Good luck with resolving the issue you encountered, and thanks for providing feedback on the stop code docs.

closed 2020-12-08 by DOMARS