MicrosoftDocs/windows-driver-docs #2236: Could you guys make the guide page more like a guide?

stang2019, 2020-05-06:
Say, each page has prev and next button, people just click those two buttons to navigate, I believe you got my points

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 862ee2d3-3052-822c-81e3-da9abca04113
* Version Independent ID: ea4573ba-40c8-b9fc-931e-247a2d489455
* Content: [AVStream Minidrivers Design Guide - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/stream/avstream-minidrivers-design-guide#feedback)
* Content Source: [windows-driver-docs-pr/stream/avstream-minidrivers-design-guide.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/stream/avstream-minidrivers-design-guide.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

RAJU2529, 2020-05-07:
@stang2019.Ya i understood your point.

barrygolden, 2020-11-21:
Thanks for reaching out to the Windows Driver Developer Documentation team. Unfortunately, we don’t have the resources to offer individual technical advice to the broad array of customers who use the driver documentation.

You may have success posting specific questions in the following forums, where other driver developers and Microsoft engineers can be found. The community there is quite responsive.

[MSDN Windows Hardware WDK and Driver Development forum](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)
[NTDEV forum at OSR Online](https://www.osronline.com/showlists.cfm?list=ntdev)

closed 2020-11-21 by barrygolden