MicrosoftDocs/windows-driver-docs #2243: Kernel debugger config fails because can't find DebugCRT version

cubisttriangle, 2020-05-08:
During 'Configure kernel debugger settings...', system cannot find 'C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Redist\MSVC\14.25.28508\debug_nonredist\X64\Microsoft.VC141.DebugCRT', because it doesn't exist, however '...VC142.DebugCRT' does exist.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ce050ea3-d61e-830c-3038-f725eb4c29e1
* Version Independent ID: eb7fc65e-bcfe-dc86-3f8c-336d20eb0c86
* Content: [Provision a computer for driver deployment and testing (WDK 10) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/gettingstarted/provision-a-target-computer-wdk-8-1#feedback)
* Content Source: [windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/gettingstarted/provision-a-target-computer-wdk-8-1.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

cubisttriangle, 2020-05-08:
To get past this, I has to open a cmd terminal as admin and create a link with the old version to the new version:

> cd C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Redist\MSVC\14.25.28508\debug_nonredist
> MKLINK /J x86\Microsoft.VC141.DebugCRT x86\Microsoft.VC142.DebugCRT
> MKLINK /J x64\Microsoft.VC141.DebugCRT x64\Microsoft.VC142.DebugCRT

DOMARS, 2020-05-08:
@cubisttriangle - Sounds like you found a solution. For these types of WDK issues, you may have success posting your question in the WDK forum, where other driver developers and Microsoft engineers can be found. 
•	[Windows Hardware WDK and Driver Development](https://social.msdn.microsoft.com/Forums/windowsdesktop/en-US/home?forum=wdk)

The first pinned article in the forum is from the WDK team, and they describe a similar workaround. Good luck with your driver development project.

closed 2020-05-08 by DOMARS