MicrosoftDocs/windows-driver-docs #2247: HPGL/2 for win10

N1FNE, 2020-05-09:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: e6554605-600d-3ff1-2f1f-9c327f91a7fa
* Version Independent ID: 37e1baa0-8c61-2616-e52e-a4907b074f90
* Content: [Plotter Driver Components - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/print/plotter-driver-components)
* Content Source: [windows-driver-docs-pr/print/plotter-driver-components.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/print/plotter-driver-components.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2020-05-10:
@N1FNE, can you expand on your comment title for this issue?

Thanks!

barrygolden, 2020-05-18:
Closing due to no information in feedback text.

closed 2020-05-18 by barrygolden