MicrosoftDocs/windows-driver-docs #2253: SSL.com Signing Certificate does not chain to a Microsoft Root Certificate

wbrosche, 2020-05-12:
D:\Holt\mPCIe2130Win10\driver\signed>signtool verify /v /kp holt1553.sys

Verifying: holt1553.sys

Signature Index: 0 (Primary Signature)
Hash of file (sha256): 9989982146D4CA090F3769B9B8FC1061FF2B132FE404176B137AC93C98DEA827

Signing Certificate Chain:
    Issued to: SSL.com EV Root Certification Authority RSA R2
    Issued by: SSL.com EV Root Certification Authority RSA R2
    Expires:   Fri May 30 13:14:37 2042
    SHA1 hash: 743AF0529BD032A0F44A83CDD4BAA97B7C2EC49A

        Issued to: SSL.com EV Code Signing Intermediate CA RSA R3
        Issued by: SSL.com EV Root Certification Authority RSA R2
        Expires:   Wed Mar 22 12:44:23 2034
        SHA1 hash: D2953DBA95086FEB5805BEFC41283CA64C397DF5

            Issued to: W. G. Holt Inc.
            Issued by: SSL.com EV Code Signing Intermediate CA RSA R3
            Expires:   Tue Mar 23 16:11:44 2021
            SHA1 hash: 47C1160D1AFC7A9BD55CE1F8E44EEB58213714F8

The signature is timestamped: Tue May 12 08:10:01 2020
Timestamp Verified by:
    Issued to: SSL.com Root Certification Authority RSA
    Issued by: SSL.com Root Certification Authority RSA
    Expires:   Tue Feb 12 12:39:39 2041
    SHA1 hash: B7AB3308D1EA4477BA1480125A6FBDA936490CBB

        Issued to: SSL.com Timestamping Issuing RSA CA R1
        Issued by: SSL.com Root Certification Authority RSA
        Expires:   Sun Nov 12 13:50:05 2034
        SHA1 hash: 84DC2F563A1AAA690C468F3B56EF2D63AEC7CA39

            Issued to: SSL.com Timestamping Unit 2019
            Issued by: SSL.com Timestamping Issuing RSA CA R1
            Expires:   Sat Nov 10 13:58:26 2029
            SHA1 hash: 986D74F120B5B18ADC7CE7B55294BDF6D8466950

SignTool Error: Signing Cert does not chain to a Microsoft Root Cert.

Number of files successfully Verified: 0
Number of warnings: 0
Number of errors: 1

RAJU2529, 2020-05-13:
@wbrosche. could you share this file  **holt1553.sys** here

EliotSeattle, 2020-05-13:
Hi @wbrosche 

This feedback forum gets sent to the Windows driver documentation team. We use it to track ***documentation*** issues. *The Hardware Dev Center Support team* has the right people to help investigate a customer problem with signing and submissions, like your issue. To open a ticket with the support team:

1. **First**, sign into your hardware dev center account, and *then*...
2. Go to [Hardware Dashboard FAQ](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq), where you'll find this passage: 
 
**How do I contact Partner Center Support?**
If you are having problems accessing the Dashboard or need Dashboard Support please open a support ticket here: [https://developer.microsoft.com/windows/support](https://developer.microsoft.com/windows/support). Select **Contact us**, **Dashboard issue**, and then **Hardware submissions & signing (all OS version)** from the dropdown menu.

It's important that you sign into your account first.

If support, working with the dashboard developers, finds that something needs to change in the docs, we'll then update that here. 

I hope this helps.

Eliot

closed 2020-05-13 by EliotSeattle