MicrosoftDocs/windows-driver-docs #2265: What does the sleep button do?

PoolloverNathan, 2020-05-20:
I know that the sleep button puts your computer in S{1..3}, but which one?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 9e17d9b3-9e6b-df00-e249-38a0cd5e1099
* Version Independent ID: 83d9bce2-8b16-812f-d587-61527d4eba50
* Content: [System Sleeping States - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/system-sleeping-states#feedback)
* Content Source: [windows-driver-docs-pr/kernel/system-sleeping-states.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/system-sleeping-states.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2020-05-21:
@tedhudek .  We can't give real answer without  practical knowledge.  I knew that some desktop motherboards have sleep button policies are enabled like S1 or S2 or S3 or S4.
Without testing in all modes, real solution is not possible.
Keep waiting for experienced users replies.

PoolloverNathan, 2020-05-21:
Okay.

Sent with [ProtonMail](https://protonmail.com) Secure Email.

‐‐‐‐‐‐‐ Original Message ‐‐‐‐‐‐‐

tedhudek, 2020-07-30:
Thanks for the question.

Typically the answer is S3.  I've just added some detail to https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/system-sleeping-states that should be live in a few minutes. You'll also see a link to the commit above if you want to see what changed.

Hope this helps!

closed 2020-07-30 by tedhudek