MicrosoftDocs/windows-driver-docs #2270: How to sign UWP app with SHA256 cert?

fman028, 2020-05-20:
I'm trying to get a SCCD to work with a blank UWP app.

It looks like be default the TemporaryKey.pfx generated for a blank UWP app is SHA1.

So, how do I "Otherwise, use a SHA256 cert to sign your app."?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 6d835c91-424a-a096-ea00-f1dd7c0bc0c4
* Version Independent ID: 1e73f683-6a1e-6794-8831-b6e6b170fa88
* Content: [Hardware Support App (HSA) Steps for App Developers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devapps/hardware-support-app--hsa--steps-for-app-developers#feedback)
* Content Source: [windows-driver-docs-pr/devapps/hardware-support-app--hsa--steps-for-app-developers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devapps/hardware-support-app--hsa--steps-for-app-developers.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

smadasam, 2020-10-27:
@fman028 I am not exactly sure what you are asking, but generally, for the authorized entity elements, the signature hash it is looking for is what signed the package.  

You can use certutil on the [package_root]\AppxMetadata\CodeIntegrity.cat file in your appx package to find the sha256 version of the signature.  The output will look something like this:

> certutil.exe codeintegrity.cat 
...
Cert Hash(sha256): 7d468d8f6311fc17ccd72ec2ad854ad7b3ee0a00b231f24d66c5505768d722e4 
Signature Hash: **06633c347e5ffd6763f4329ca7f4c8d867b6b1e087b515caded7000000000000** 
----------------  End Nesting Level 1  ---------------- 
No CRLs CertUtil: -dump command completed successfully.

If you are building the package in VS, then it is signed with your developer cert.  If you install your app with the VS ps script, it does some nice things for you like installing your developer cert.

Store packages are signed with the store cert.  In this case, when you ask HSA review to sign your SCCD, they will automatically add a new authorized entity with the apps PFN and the store's cert's signature hash for you.

How the app is installed with an SCCD  depends on the SKU of windows, but on Desktop based SKUs, you can just enable developer mode, and the extra certificate checks will be skipped.  This is a good place to get started.

If you are deploying for some other SKUs, like IoT or HoloLense, you will have to manually install your developer cert into the cert store otherwise the certificate will not be trusted by the OS.

Regardless, the XML needs to be valid.  For example, you still need to have some kind of valid base64 value in the catalog element even if the SCCD is not MS signed like "0000" is fine, or the XML will not validate against the XSD (XML schema).  The schema files are published, so you should do your own validation before trying to install the package.  If the package won't install in developer mode, it is probably because it violates the schema.

You can see the reason why a package failed to install in different ways depending on how you are installing it, but in the event viewer should have it in case you missed it.

But, to start working with your SCCD, you

fman028, 2020-10-27:
Thanks for the detailed response @smadasam!  I was deploying to a WCOS vm, which is a bit more finicky than Windows10.

closed 2020-10-27 by tedhudek