MicrosoftDocs/windows-driver-docs #2272: MASM integer literal radix suffix documentation lacking

EugeneKramer, 2020-05-21:
The documentation for MASM integer literals/constants radix suffixes is lacking. 

The only one mentioned in the documentation is _h_: "You can also specify hexadecimal numbers by adding an h after the number."

Supported radix suffixes include:

- h - hexadecimal
- o/q - octal
- d/t - decimal
- b/y - binary
- r - encoded real

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 4c2678ec-32bd-a941-e210-2d75fd4e04a7
* Version Independent ID: c1ebc3a4-3a7a-ba2b-6a3f-25d6baef5e3a
* Content: [MASM Numbers and Operators - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/masm-numbers-and-operators)
* Content Source: [windows-driver-docs-pr/debugger/masm-numbers-and-operators.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/masm-numbers-and-operators.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-07-21:
@EugeneKramer - I was not able to use the additional suffixes. Can you provide a simple example that shows their use? Here is what I saw. 

```
0:008> .expr /s masm
Current expression evaluator: MASM - Microsoft Assembler expressions

0:008> n 
base is 10

0:008> ? @$TEB+ FFh
Evaluate expression: 548456567039 = 0000007f`b28f10ff

0:008> ? @rax + 11h
Evaluate expression: 548456566801 = 0000007f`b28f1011

0:008> ? @rax + 11d
Syntax error at '11d'

0:008> ? @$TEB+ 11b
Syntax error at '11b'

0:008> ? @$TEB+ FF
Couldn't resolve error at 'FF'

0:008> ? @$TEB+ 0xFF
Evaluate expression: 548456567039 = 0000007f`b28f10ff

0:008> ? 11 + 11
Evaluate expression: 22 = 00000000`00000016

0:008> ? 11 + 11h
Evaluate expression: 28 = 00000000`0000001c

0:008> ? 11 + 11b
Syntax error at '11b'

```

There sometimes seems to be some some ideocycronoies to the parser.

KalleOlaviNiemitalo, 2020-08-29:
Perhaps the title "MASM Numbers and Operators" gives the false impression that the page also documents MASM itself, rather than only MASM expressions in the debugger. For MASM, most of the radix suffixes are listed under `floatNumber` or `radixOverride` in [Microsoft Macro Assembler BNF Grammar](https://docs.microsoft.com/en-us/cpp/assembler/masm/masm-bnf-grammar?view=vs-2019). Does MASM actually support `b` and `d`, which are not listed there?

EugeneKramer, 2020-08-29:
@KalleOlaviNiemitalo is correct. The debugger was wrongly linked. `B(b)` and `D(d)` may be covered under the `hexdigit` portion of the BNF tree. 

There is [Intel documentation](http://pong.tamu.edu/~baum/docs/intel70/ref/asm_lan_lx.htm#cnst_msm.htm) for supporting D(d) and B(b).

Radix | Radix Indicator Suffix|
|------|------------------------|
|Decimal | D (d), or T (t) when the current radix is hex|
|Hexadecimal | H (h)|
|Octal | O (o) or Q (q)|
|Binary | B (b), or Y (y) when the current radix is hex|
|Encoded real| R(r)|

A MASM radix page could not be found in the docs...perhaps it could be created. 

This issue can be closed as it was incorrectly filed.

KalleOlaviNiemitalo, 2020-08-31:
Could be added as part of [.RADIX](https://docs.microsoft.com/cpp/assembler/masm/dot-radix?view=vs-2019) documentation, or at least linked from there.

DOMARS, 2021-01-27:
@EugeneKramer and @KalleOlaviNiemitalo - Thanks for the input on this topic. I added new introductory text and updated some sub topic titles to better highlight that this content is for the implementation of the MASM expression syntax in the Windows debugging tools, such as WinDbg. If you think there is a need to clarify the use of radix in the MASM docs, the best route would be open a git issue on the specific topic. If you have specific text that should be added, consider creating a pull request with the doc updates for those topics. Thanks for your interest in the documentation.

closed 2021-01-27 by DOMARS