MicrosoftDocs/windows-driver-docs #2274: "Desktop" section misalignment and missing "Windows Driver" target.

dluzak, 2020-05-21:
It seems like the "| |Desktop|A driver binary for Windows 10 for desktop editions or Windows Server 2016 can call this DDI.|" section from the end of the article should be a part of the table with Target Platforms.
Another thing is that in the latest release of WDK there is also "Windows Driver" target, but I can't find any documentation for it. Could you please describe it here?
Thanks!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c3764139-c203-f295-1591-e8c67ee0eb94
* Version Independent ID: 8128169c-8bab-008b-a7e4-bd647f4c02d2
* Content: [Target platform on driver reference pages - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/target-platforms#feedback)
* Content Source: [windows-driver-docs-pr/develop/target-platforms.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/target-platforms.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-05-21:
Hi @dluzak, thanks for the feedback.  Check out https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/getting-started-with-windows-drivers for info on Windows Drivers.

tedhudek, 2020-05-21:
Just pushed the table fix, should appear soon.   Thanks again for the feedback!

closed 2020-05-21 by tedhudek