MicrosoftDocs/windows-driver-docs #2275: Link Typo in First Paragraph

druske, 2020-05-21:
There is a link to a page on using device installation functions whose text is "Using Device Interface Functions"; I believe the intended text should be "Using Device Installation Functions."

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f084bfb2-4458-d1e5-94ad-71eb2b72e0c9
* Version Independent ID: ae73936c-ea59-8ba3-4492-60032c48652c
* Content: [Using Device Interfaces - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/wdf/using-device-interfaces)
* Content Source: [windows-driver-docs-pr/wdf/using-device-interfaces.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/wdf/using-device-interfaces.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2020-05-21:
Created PR #2277 .

tedhudek, 2020-05-21:
Thanks @druske @RAJU2529

closed 2020-05-21 by tedhudek