MicrosoftDocs/windows-driver-docs #2281: what is a base INF

zhengxi0922, 2020-05-26:
[Enter feedback here]
what is a base INF and what is it for?
Does component inf refer to the base INF?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 61e5e417-647c-0c3a-9c25-83a7be9efe3a
* Version Independent ID: 6ccb6634-97cb-b66f-3ca5-0e58218abdc0
* Content: [Using an Extension INF File - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-an-extension-inf-file)
* Content Source: [windows-driver-docs-pr/install/using-an-extension-inf-file.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/using-an-extension-inf-file.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2020-06-02:
@zhengxi0922 . base inf is the standard information that should contains in .inf files. I seen many .inf files, some device manufacturers include many configurations additional to the standard configurations. Also some.inf files contains very less configurations

tedhudek, 2020-06-02:
Thanks for the feedback.  We're currently working on additional documentation to clarify usage of base and extension INFs. I'll update here when that becomes available.

tedhudek, 2020-07-22:
Sorry for the delay - here are some links to the latest info on base and extension INFs:

https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/dch-principles-best-practices
https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/dch-example
https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-an-extension-inf-file
https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-an-extension-inf-file-template

closed 2020-07-22 by tedhudek