MicrosoftDocs/windows-driver-docs #2285: Remove backslashes in command names?

CedricHerzog, 2020-05-27:
A backslash is added before each \* if the command name ends with \*:
- ".sympath srv\\\*" instead of ".sympath srv\*"  (4)
- "x nt!*CreateProcess\\\*" instead of "x nt!*CreateProcess\*" (9)
- "bc \\\*" instead of "bc \*" (13)

Shouldn't it be removed?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ae1f7f88-1efe-7e6c-ed65-5a5c0c4a9edc
* Version Independent ID: 857e3b38-a886-8bcf-8878-37fa19c7d228
* Content: [Getting Started with WinDbg (Kernel-Mode) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/getting-started-with-windbg--kernel-mode-)
* Content Source: [windows-driver-docs-pr/debugger/getting-started-with-windbg--kernel-mode-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/getting-started-with-windbg--kernel-mode-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

EliotSeattle, 2020-05-28:
Thanks for bringing this to our attention, @CedricHerzog  
I made the changes in the source. The file is being reviewed before it merges.
Eliot

RAJU2529, 2020-06-02:
I Will check all the links once again'

DOMARS, 2020-06-02:
@CedricHerzog - Thanks for noticing these command escape problems and filing this issue. The PR should have fixed all them so closing this issue. Thanks @EliotSeattle for the PR!

closed 2020-06-02 by DOMARS