MicrosoftDocs/windows-driver-docs #2286: useless

deep2002n, 2020-05-27:
[Enter feedback here]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 0aff8a86-8549-bb27-effc-34d05b05fa06
* Version Independent ID: aa42165e-1a7f-5c1c-c19b-65e41754b318
* Content: [PwrTest Sleep Scenario - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/devtest/pwrtest-sleep-scenario)
* Content Source: [windows-driver-docs-pr/devtest/pwrtest-sleep-scenario.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/devtest/pwrtest-sleep-scenario.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-05-27:
Hi @deep2002n 
Appreciate that you took the time to create the feedback. Can you give me something to go on in order to see if I can make the content better? What was missing that you expected to find? What is the in that topic that isn't working for you?
Thanks,
Eliot

RAJU2529, 2020-05-27:
@EliotSeattle . Could you tell me why this is useless. I found those are useful. Commamds for testing laptop battery how much time giving life to work.

closed 2020-07-02 by EliotSeattle

deep2002n, 2020-07-02:
Hi,I am not related to this work (or any other work) on GitHub, it seems the comment was put by mistake
Please delete it, I checked but could not find option to delete.
I regret the in-convenience .
Deep Narayan

RAJU2529, 2020-07-02:
@deep2002n . Click three dots, rename the title and click save

EliotSeattle, 2020-07-02:
This issue is closed. No need for further action.
And no worries, @deep2002n. I just expect to see actionable feedback from you next time. ;-)
Thanks,
Eliot