MicrosoftDocs/windows-driver-docs #2288: UMDF 2 vs KMDF

druske, 2020-05-27:
This page contains a table comparing Windows Drivers with Windows Desktop Drivers. One difference called out is whether WHCP certification is required. Is WHCP certification required only for kernel (KMDF) drivers, or is it required for user-mode (UMDF) drivers as well? Clarification would be helpful.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2c5fa76b-6c2e-8f5e-7e96-ceb1c726842b
* Version Independent ID: 48a29c52-9907-9b1a-1dc2-3365c2d021d0
* Content: [Getting Started with Windows Drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/getting-started-with-windows-drivers)
* Content Source: [windows-driver-docs-pr/develop/getting-started-with-windows-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/getting-started-with-windows-drivers.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

selerner, 2020-05-29:
Hi! This is Sebastian from MSFT, WHCP certification for Windows Drivers is required for KMDF and UMDF drivers.

EliotSeattle, 2020-05-29:
Edited the last bullet in the list to read:

- Certified with [Windows Hardware Compatibility Program Certification Process](https://docs.microsoft.com/windows-hardware/design/compatibility/whcp-certification-process) using the [Hardware Lab Kit](https://docs.microsoft.com/windows-hardware/test/hlk/). Note that the Windows Hardware Compatibility Program Certification Process requirements apply to both KMDF and UMDF drivers.

closed 2020-05-29 by EliotSeattle