MicrosoftDocs/windows-driver-docs #2289: The certificate used for submission signing must be registered

emuzychenko, 2020-05-28:
If the certificate used for signing the submission, is not registered in Partner Center (added to the dashboard by signing/uploading a signable file), the submission fails with "SignatureValidationFailed", with no further explanation.

Please add this peculiarity to the documentation and make the error message more informative.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ed287852-04b7-1569-fd51-2640cfa24f57
* Version Independent ID: a8ce732f-5b41-f1ce-e37d-a7c2794546a4
* Content: [Attestation signing a kernel driver for public release - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/attestation-signing-a-kernel-driver-for-public-release)
* Content Source: [windows-driver-docs-pr/dashboard/attestation-signing-a-kernel-driver-for-public-release.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/attestation-signing-a-kernel-driver-for-public-release.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

tedhudek, 2020-10-28:
Hi @emuzychenko thanks for the feedback and apologies for the delay.  I just pushed d2e384cb2 with updates for your request. The changes should appear on the live site momentarily.

I also passed along your feedback on the error message to the right people.

closed 2020-10-28 by tedhudek

TechnikEmpire, 2021-02-18:
I'm not seeing this update and googling brought me to this bug ticket, which saved me pulling my hair out. Maybe I'm blind or lazy but I ctrl+f looking for info in doc pages and "signable" or "manage certificates" or "certificates" yields nothing in this resource.

tedhudek, 2021-02-18:
@TechnikEmpire did you try the link to the commit in my comment above?  That still works for me - let me know if that doesn't help

TechnikEmpire, 2021-02-19:
@tedhudek ah I see, my bad. If I might offer a little feedback, I didn't even bother to check the link about acquiring an EV cert, because I've already done that. So, in my mind at least, I wouldn't revisit that page. I ran into this issue because I renewed my EV cert yesterday and was puzzled by why the hardware portal was rejecting my uploads. Anyway just my 2 cents, thanks for pointing out the fix to me.