MicrosoftDocs/windows-driver-docs #2290: Information regarding _PR2 and _PR3 is incorrect

rgough71, 2020-05-28:
[Enter feedback here]

Information regarding Power Resource descriptions is incorrect.
Table shows that requirements for D3(hot) are described by _PR2, and D3(cold) are described in _PR3. This is incorrect
It also correctly shows that _PR2 shows requirements for D2.

Requirements for D3(hot) are provided by _PR3.
There are no power requirements for D3(cold) using _PRx objects.

Also, this page refers to ACPI 5.0. The latest version of the ACPI specification is 6.3. The link is still correct.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b168f14f-59e0-9fe6-d600-6e06621e4b77
* Version Independent ID: 79f32da9-5815-a833-7683-50016bd272f3
* Content: [Device power management - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/device-power-management)
* Content Source: [windows-driver-docs-pr/bringup/device-power-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/device-power-management.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2021-01-29:
@rgough71, thanks for your detailed feedback.

I'm adding the information you've provided to an internal work item to correct and update the Device power management topic.

barrygolden, 2021-03-28:
Topic was updated under the following PRs:

https://github.com/MicrosoftDocs/windows-driver-docs/pull/2300
https://github.com/MicrosoftDocs/windows-driver-docs/pull/2302

closed 2021-03-28 by barrygolden