MicrosoftDocs/windows-driver-docs #2292: I cannot find Windows 10 SDK (10.0.19041.1)  by using the Modify button in Visual Studio install.

tomxue, 2020-05-29:
After reading this:
https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md

I cannot find Windows 10 SDK (10.0.19041.1)  by using the Modify button in Visual Studio install. Why?
My Windows is just updated to Version 1909, the latest one. 

And the link above does not say cleayly, if Windows 10 SDK (10.0.19041.0) is installed, is it enough for driver development? Or I must install 10.0.19041.1? But where is it?

I have to say, develop a WDK driver is so frustrated, I was blocked at the beggining by these kinds of fuzzy issues.

RAJU2529, 2020-05-29:
@tomxue .     windows 10 SDK 19041 is updated version from 1909.
Moreover you can develop compatible drivers for 19041to windows 10 v2004 update.
And 18362 SDK can be used to develop drivers for windows 10 1903 and 1909.
.

tomxue, 2020-05-29:
@RAJU2529 Thank you for your reply!

(1)
For this link: https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/download-the-wdk.md, It says:
```
To select the correct SDK:

In Visual Studio Installer, on the Individual components tab, search for Windows 10 SDK (10.0.19041.0), select this version and continue with install.

If you already have Visual Studio 2019 installed, you can install the Windows 10 SDK (10.0.19041.1) by using the Modify button in Visual Studio install.
```

But I tries a lot of times, and **can not** install the Windows 10 SDK (10.0.19041.1) by using the Modify button in Visual Studio install.

This wastes my time and makes me feel frustrated. Please confirm it, if yes, could you please modify it?

(2)
For this Microsoft code:
https://github.com/Microsoft/Windows-driver-samples

Related document:
https://docs.microsoft.com/en-us/windows-hardware/drivers/download-the-wdk

Like you said, it gives hint: Note that the version of the SDK that is compatible with the WDK for Windows 10, version 2004 may not be the default SDK. 

But when I install 10.0.18362.1 version WDK, I met a lof of strange issues and I spent nearly 1 week, still cannot even run the debug process on the target computer.

Microsoft's driver development process seems so difficult to set up. Why?

RAJU2529, 2020-05-29:
Just now I seen windows 10 SDK 19041 inside the visual studio 2019 community edition.  Go to menu in visual studio, click available, then click on individual components, scroll down Bottom of the page,

tomxue, 2020-05-29:
[image]

```
To select the correct SDK:

In Visual Studio Installer, on the Individual components tab, search for Windows 10 SDK (10.0.19041.0), select this version and continue with install.

If you already have Visual Studio 2019 installed, you can install the Windows 10 SDK (10.0.19041.1) by using the Modify button in Visual Studio install.
```

The key here is 10.0.19041.1
The picture you show confirms that **can *not* install the Windows 10 SDK (10.0.19041.1) by using the Modify button in Visual Studio install.**

RAJU2529, 2020-05-29:
@tomxue .  remove the old versions of all SDK

tomxue, 2020-05-29:
[image]

I remove the old versions of all SDK and still can not find 19041.1
Why?

RAJU2529, 2020-05-29:
@tomxue. Might be some other dependency files or  .net version is missing. 
i am downloading here  windows 10 SDK 19041, once complete means, I will copy  to other location for future installations

tomxue, 2020-05-29:
I do not fully understand you.
If you solve it, please let me know. Thanks!

RAJU2529, 2020-05-29:
@tomxue . Jsut now I installed it .

tomxue, 2020-06-02:
@RAJU2529 But how do you install 19041.1? By  remove the old versions of all SDK?
I did so and still could not find 19041.1.

RAJU2529, 2020-06-02:
@tomxue , there is an another way to install sdk 19041 without using visual studio 2019. 
i am attached  web installer setup.exe file, download and run with admin rights, the setup file will download all required files, you can save those files for future installation.

[winsdksetup.zip](https://github.com/MicrosoftDocs/windows-driver-docs/files/4715211/winsdksetup.zip)

tomxue, 2020-06-02:
Thank you for your help! The file you provide is workable to install sdk 19041.1.
And also I am right: sdk 19041.1 installation method of official document is not right. Hope someone could modify it.

EliotSeattle, 2020-08-20:
We've updated the steps on the page to indicate that Visual Studio does, indeed, end up installing the correct version, though the UI there doesn't explicitly say so.
Thanks for the Issue!

closed 2020-08-20 by EliotSeattle

WCHIHO, 2021-11-10:
> @tomxue , there is an another way to install sdk 19041 without using visual studio 2019. i attached web installer setup.exe file, download and run with admin rights, the setup file will download all required files, you can save those files for future installation.
> 
> [winsdksetup.zip](https://github.com/MicrosoftDocs/windows-driver-docs/files/4715211/winsdksetup.zip)

The file I tried to download is Windows 10 SDK (10.0.19043,0) which is not listed as an individual component to install in Microsoft Visual Studio Community 2019. I received this error message whilst "PIP install qpsolvers: was being executed:

    -- Selecting Windows SDK version  to target Windows 10.0.19043.
    CMake Error at CMakeLists.txt:4 (project):
      Failed to run MSBuild command:

        MSBuild.exe

      to get the value of VCTargetsPath:

        The system cannot find the file specified
============================================

So, can you provide a link to download  Windows 10.0.19043? Where should it be download to?

By the way, is there another way to solve this issue of error :c\qdldl/include/qdldl.h(5): fatal error C1083: Cannot open include file: 'qdldl_types.h': No such file or directory.