MicrosoftDocs/windows-driver-docs #2297: Errata 41600?

druske, 2020-06-01:
There's an issue at the bottom of the page regarding a link error with Static Driver Verifier. The proposed solution is "Please use errata 41600 for this issue."

What is "errata 41600" and why is there no link to it? The cause of the error is not even hinted at. This is about as cryptic as documentation can be, and even a Google search isn't turning up anything useful yet.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3cce6b03-d7a7-bd33-9e2f-9a7df796c315
* Version Independent ID: d0e890b7-a43c-861a-1e6f-2205a9892409
* Content: [Static Driver Verifier Known Issues - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/develop/static-driver-verifier-known-issues)
* Content Source: [windows-driver-docs-pr/develop/static-driver-verifier-known-issues.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/static-driver-verifier-known-issues.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @selerner
* Microsoft Alias: **selerner**

RAJU2529, 2020-06-02:
@druske . Might be that is new error, in coming days , error solution will be included

DOMARS, 2020-06-04:
@druske - I checked with the HLK team and they clarified that the errata note is for partners working on WHCP certification. So I updated the wording to indicate that, and also that this is a known issue. As not everyone is using SDV to work on certification, I understand how the original wording was confusing.

closed 2020-06-04 by DOMARS