MicrosoftDocs/windows-driver-docs #2298: Articulate that D3Cold requires _S0W(4)

tzimmer1, 2020-06-02:
[Enter feedback here]
Per other public docs and Intel testing, please articulate on this page in the “_SxW” section that _S0W(4) is a requirement for D3Cold regardless of whether or not the device supports wake.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b168f14f-59e0-9fe6-d600-6e06621e4b77
* Version Independent ID: 79f32da9-5815-a833-7683-50016bd272f3
* Content: [Device power management - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/device-power-management)
* Content Source: [windows-driver-docs-pr/bringup/device-power-management.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/device-power-management.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

barrygolden, 2021-01-29:
Adding this feedback to an internal work item to update the Device power management topic.

barrygolden, 2021-03-28:
Updated topic per feedback above:

https://github.com/MicrosoftDocs/windows-driver-docs-pr/pull/1026

closed 2021-03-28 by barrygolden