MicrosoftDocs/windows-driver-docs #2312: Where does WexTestClass.h come from?

DepthDeluxe, 2020-06-09:
This documentation is very unclear on how to set up a Visual Studio project that can create TAEF tests.  The docs provide a useful example test but there is no example on how to set up the project.  I created a "C++ Unit Test Project" in Visual Studio 2019 and attempted to `#include <WexTestClass.h>` but did not find it.

Can you provide either an example project or step-by-step instructions on how to get simple dummy tests started?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 19dbeee9-fda6-b1bf-6973-7f07d3b69b0e
* Version Independent ID: 3e5b4cd6-c034-33a0-13dd-bff67840f24f
* Content: [Authoring Tests in C++ - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/taef/authoring-tests-in-c--)
* Content Source: [windows-driver-docs-pr/taef/authoring-tests-in-c--.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/taef/authoring-tests-in-c--.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-06-09:
Thanks for the Issue, @DepthDeluxe 
You can find WexTestClass.h in this location: 

%\Program Files (x86)\Windows Kits\10\Testing\Development\inc

I've updated the doc to indicate this.
Cheers,
Eliot

closed 2020-06-09 by EliotSeattle

DepthDeluxe, 2020-06-09:
@EliotSeattle, thanks for the quick update.  I have another issue open #2313 which is related to this issue as well.  In the current documentation, it is very unclear how to get started writing tests for Kernel drivers.  I'd appreciate it if you could take a look at that issue as well.

EliotSeattle, 2020-06-09:
@DepthDeluxe, this one was simple for me. The other one is outside my expertise, but the Issue is assigned to one of our crackerjack writers. Your request on that one is more involved, so it will likely take a little more time, especially depending on the writer's workload.
Cheers,
Eliot
@tedhudek