MicrosoftDocs/windows-driver-docs #2333: lost link

mrmsbarnes, 2020-06-19:
first paragraph on page, first link https://docs.microsoft.com/en-us/windows/desktop/api/mmreg/ns-mmreg-twaveformatex is not found.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: bdfa300c-011a-75dc-2a8a-8d896f794f30
* Version Independent ID: 11d4fdcf-43a2-853d-e9c3-a8f8234f51f5
* Content: [Wave and DirectSound Components - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/wave-and-directsound-components)
* Content Source: [windows-driver-docs-pr/audio/wave-and-directsound-components.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/wave-and-directsound-components.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

RAJU2529, 2020-06-19:
@DOMARS , Issue is conformed by me also, after checking the link.

EliotSeattle, 2020-06-19:
Fixed in PR #2334.

closed 2020-06-19 by EliotSeattle