MicrosoftDocs/windows-driver-docs #2335: DefaultInstall not listed in invalid sections in a universal INF file

NextTurn, 2020-06-20:
> **Note**  If you are building a universal or mobile driver package, this section is not valid and should not be used.  Instead, use only the [**INF Manufacturer Section**](inf-manufacturer-section.md).

The *INF DefaultInstall Section* page says it's invalid in universal driver packages. If that's the case it should be listed in [the *Which INF sections are invalid in a universal INF file?* section of the *Using a Universal INF File* page](https://docs.microsoft.com/windows-hardware/drivers/install/using-a-universal-inf-file#which-inf-sections-are-invalid-in-a-universal-inf-file).

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d59dc53b-91b7-633e-6186-6715185bc886
* Version Independent ID: 3cbaf14f-56ee-0293-76dd-6914118a5240
* Content: [INF DefaultInstall Section - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-defaultinstall-section)
* Content Source: [windows-driver-docs-pr/install/inf-defaultinstall-section.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-defaultinstall-section.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-06-22:
It's valid if it has an architecture decoration, i.e. [DefaultInstall.NTAMD64]

I will add this to the doc.  Thx for pointing it out

tedhudek, 2020-07-22:
Thanks again for this feedback @NextTurn 

I just pushed a fix. Should be live on the site in a few minutes. You'll also get a link above with the commit if you want to see what changed

closed 2020-07-22 by tedhudek