MicrosoftDocs/windows-driver-docs #2337: what is the differences between compatible ID and hardware ID

zhengxi0922, 2020-06-22:
[Enter feedback here]

what is the differences between compatible ID and hardware ID?
since both IDs have the same format, why do we need compatible IDs?
and what about the priority?
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d01ed123-4144-3845-af3d-e5013ce8cc41
* Version Independent ID: 50b0c5f8-7190-786a-3cde-f7501d051d67
* Content: [Compatible ID - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/compatible-ids)
* Content Source: [windows-driver-docs-pr/install/compatible-ids.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/compatible-ids.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

tedhudek, 2020-06-22:
If Windows cannot locate an INF file that matches one of a device's hardware IDs, it uses compatible IDs to locate an INF file.

A compatible ID is usually less specific, in order to customize a setting across a broader set of devices.

Also see https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-custom-hardware-ids-and-compatible-ids

closed 2020-06-22 by tedhudek

zhengxi0922, 2020-06-23:
OK
But still, why inf creator don't use the hardware id at the first time?
"A compatible ID is a vendor-defined...."so who is the vendor?
e.g. we have a touchpad device, so the compatible ids are created by the touchpad supplier?

zlockard, 2020-06-23:
The vendor would be whoever is making the touchpad.  The general way to think about hardware or compatible IDs is that they're a contract between the device and the driver.  If a driver installs on a hardware ID, it is fulfilling a contract to provide full functionality to a device.  A compatible ID is a contract that says that at least basic functionality is implemented.  A reason to do this might be so that a generic driver that matches compatible IDs might work on a large number of devices, for example a lot of the driver built into Windows match compatible IDs.

tedhudek, 2020-06-23:
Thanks @zlockard I added this info to the bottom of https://docs.microsoft.com/en-us/windows-hardware/drivers/install/using-custom-hardware-ids-and-compatible-ids

zhengxi0922, 2020-06-24:
well understood.
thank you guys.
@zlockard @tedhudek