MicrosoftDocs/windows-driver-docs #2338: First line of C++ code on the page

johan-boule, 2020-06-22:
[Enter feedback here]

Please fix the first line of C++ code on the page. It can't possibly be correct.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 385b030c-7344-9c34-97ab-9e42098ca1cf
* Version Independent ID: d6ca0333-1804-b888-74e6-f83bae813488
* Content: [Low Latency Audio - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/audio/low-latency-audio#:~:text=Before%20Windows%2010%2C%20the%20latency,processed%20data%20to%20a%20buffer.)
* Content Source: [windows-driver-docs-pr/audio/low-latency-audio.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/audio/low-latency-audio.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-06-22:
@johan-boule - Looks like that may be C# code. Will investigate.

DOMARS, 2020-06-22:
@johan-boule - I confirmed that the first code block label was incorrect and updated it to use csharp. Thanks for seeing this inconsistency/mistake and filing a docs issue.

closed 2020-06-22 by DOMARS

johan-boule, 2020-07-14:
Sorry, but this is not an acceptable modification in the context of this article. As far as I understand, the whole article is using C++ in its examples, except for this first very simple code block which is easy to translate from C# to C++. We're talking about low-latency system programming here, so people want to see C++ code as the primary language.

DOMARS, 2020-07-31:
@johan-boule - Not sure I follow. Developers calling down into the driver can use either the C# or C++. The first sample references an example C# code sample that uses low latency features using AudioGraph. I understand your preference for C++, so you might want to look at the WASAPI sample, which will likely have lower latency as mentioned in the docs.

johan-boule, 2020-08-27:
OK. I think I don't understand who's the audience of this page then.