MicrosoftDocs/windows-driver-docs #2348: 0x10 missing

MrPepka, 2020-07-04:
I came across the blue screen 0x124 where the first parameter is 0x10. Can you add a description for this parameter?
WHEA_UNCORRECTABLE_ERROR (124)
A fatal hardware error has occurred. Parameter 1 identifies the type of error
source that reported the error. Parameter 2 holds the address of the
WHEA_ERROR_RECORD structure that describes the error conditon.
Arguments:
Arg1: 0000000000000010, Error Source Type
Arg2: ffffd08957035028
Arg3: ffffd08941b10aac
Arg4: ffffd08941b1b1a0

DOMARS, 2020-12-19:
@MrPepka - I have created an internal work item to investigate updates to include additional parameters for the WHEA_UNCORRECTABLE_ERROR (124). We use input like this to understand what people are seeing, so thanks for passing on the occurrence of param 0x10 and good luck with your debugging work.

closed 2020-12-19 by DOMARS

AntonPetrochenko, 2021-05-07:
I know this is against all principles, but I'm going to **bump** this. I ran into a bugcheck 0x124 with param 1 being 0x10 and cannot debug it because there is no documentation. It goes up to 0xB as far as docs.microsoft.com is concerned.