MicrosoftDocs/windows-driver-docs #2349: Useless page

bgroicahn, 2020-07-06:
Useless page when download is not available!

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b034fd72-470b-76e3-e080-46cb688242db
* Version Independent ID: 3fde5448-65db-8332-2194-c3b301d0d4d5
* Content: [Capture and view USB traces with Microsoft Message Analyzer - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/capture-and-view-ing-usb-traces-with-microsoft-message-analyzer-)
* Content Source: [windows-driver-docs-pr/usbcon/capture-and-view-ing-usb-traces-with-microsoft-message-analyzer-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/capture-and-view-ing-usb-traces-with-microsoft-message-analyzer-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

RAJU2529, 2020-07-07:
@bgroicahn . This **Microsoft Message Analyzer tool**  has been retired from November 2019 onwards. 
According to this article  **https://docs.microsoft.com/en-us/message-analyzer/installing-and-upgrading-message-analyzer**

EliotSeattle, 2020-07-07:
Thanks for pointing this out, @bgroicahn. @RAJU2529 is correct, the tool has been retired. I am sorry that we missed adding the note to this page. I've made that change.
Eliot

closed 2020-07-07 by EliotSeattle