MicrosoftDocs/windows-driver-docs #2358: What is the name of the REG_DWORD for firmware rollback?

goxia, 2020-07-13:
The name of this key value is not mentioned, then what should it be?

Thanks.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b02e11cd-e499-0dd8-b195-17716fa82790
* Version Independent ID: a667526a-5e3f-cb38-661a-de6cb0b61d6c
* Content: [Firmware update status - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/firmware-update-status#rolling-back-firmware-updates)
* Content Source: [windows-driver-docs-pr/bringup/firmware-update-status.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/firmware-update-status.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

RAJU2529, 2020-07-13:
@goxia .
Key path :  HKLM\SYSTEM\CurrentControlSet\Control\FirmwareResources.
Value : Policy
Type : REG_DWORD
Data :  1

goxia, 2020-07-14:
> @goxia . I think the below details might be correct. But i am not sure
> Key path : HKLM\SYSTEM\CurrentControlSet\Control\FirmwareResources.
> Value : Policy
> Type : REG_DWORD
> Data : 1

Thank you a lot.

closed 2020-07-14 by goxia