MicrosoftDocs/windows-driver-docs #2359: This API is broken server-side, 500 errors

ygel, 2020-07-13:
Exploring available API for submitting drivers for attestation and retrieving signed drivers. Seems like everything is broken as of 7/12/20.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: ead14b61-61e5-d45f-d5c9-d3d4c0bcf93d
* Version Independent ID: 31c51703-a968-0a6a-bba0-bdb47619b018
* Content: [Hardware dashboard API - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/dashboard-api)
* Content Source: [windows-driver-docs-pr/dashboard/dashboard-api.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/dashboard-api.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

ygel, 2020-07-13:
POST's to https://login.microsoftonline.com{my tenant id}/oath2/token with url-encoded form (grant_type=client_credentials, client_id, client_secret, resource=https://manage.devcenter.microsoft.com) return 404. Using C# sample returns 500:
```
Hello World!
----------Initiating Create ReportTemplate API Call
----------Create ReportTemplate Initiated, response received: { "statusCode": 500, "message": "Internal server error", "activityId": "guid" }
**********************************
Bye Bye World!
```

EliotSeattle, 2020-10-23:
Hi @ygel 

I am so sorry that this sat without a reply. 

For your specific questions, Hardware Dev Center Support can investigate. To contact support and open a Hardware Dev Center support case for investigation, **first**, sign into your hardware dev center account, and **then** go to [https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq](https://docs.microsoft.com/windows-hardware/drivers/dashboard/hardware-dashboard-faq).

Find: 
 
**How do I contact Partner Center Support?**
If you are having problems accessing the Dashboard or need Dashboard Support please open a support ticket here: [https://developer.microsoft.com/windows/support](https://developer.microsoft.com/windows/support). 

Once there, select **Contact us**, **Dashboard issue**, and then **Hardware submissions & signing (all OS version)** from the dropdown menu.

Hopes this helps.
Eliot

closed 2020-10-23 by EliotSeattle