MicrosoftDocs/windows-driver-docs #2361: Screenshot is  better

RAJU2529, 2020-07-13:
I like to add screenshots of system firmware updated with related registry key path. 
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: d644fa9d-d986-55d1-b9e7-cba9020b5553
* Version Independent ID: 342bd7c0-9e7a-5d5a-6f1e-50ba7a5d968f
* Content: [ESRT table definition - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/esrt-table-definition)
* Content Source: [windows-driver-docs-pr/bringup/esrt-table-definition.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bringup/esrt-table-definition.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @barrygolden
* Microsoft Alias: **bagold**

RAJU2529, 2020-07-13:
here are the screenshots.
[image]
[image]
[image]
[image]
[image]

barrygolden, 2020-11-04:
Thanks for submitting these screenshots, @RAJU2529.  However I'm not sure of the context for these and how they would fit into this reference focused topic. These seem to be oriented more towards a walkthrough article and would need additional text explaining each step you are showing in the images. Also, for future reference, screenshots are limited to 1200 pixels in height and width.

closed 2020-11-04 by barrygolden