MicrosoftDocs/windows-driver-docs #2367: Invalid email address for accessibility API

terriblememory, 2020-07-17:
Hi. This page specifies acc_driver@microsoft.com as an address to contact for information on an accessibility API. However attempting to mail this address results in an immediate bounce with the following error:

550 5.4.1 Recipient address rejected: Access denied. AS(201806281) [DM3NAM06FT016.Eop-nam06.prod.protection.outlook.com]

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 5cf45757-811d-78f5-6396-e750aa16d41f
* Version Independent ID: 447b7330-f0d2-c6d6-0b52-2a73c65ed06c
* Content: [Roadmap for the Windows Display Driver Model (WDDM) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/roadmap-for-developing-drivers-for-the-windows-vista-display-driver-mo)
* Content Source: [windows-driver-docs-pr/display/roadmap-for-developing-drivers-for-the-windows-vista-display-driver-mo.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/roadmap-for-developing-drivers-for-the-windows-vista-display-driver-mo.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

terriblememory, 2020-07-17:
Ugh, I used the link on the wrong page... gimme a minute...

EliotSeattle, 2020-07-17:
@terriblememory

lorihollasch, 2020-07-18:
@terriblememory , I've updated the Roadmap page to be current. @EliotSeattle is still working on tracking down the accessibility email alias - stay tuned!

closed 2020-07-18 by lorihollasch