MicrosoftDocs/windows-driver-docs #2368: Invalid email address for accessibility mirror driver information

terriblememory, 2020-07-17:
This page specifies acc_driver@microsoft.com as the contact for information on accessibility mirror drivers, however mail sent to it immediately bounces with this error:

> 550 5.4.1 Recipient address rejected: Access denied. AS(201806281) [DM3NAM06FT016.Eop-nam06.prod.protection.outlook.com]

Does the address need to be updated?

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2ccfb830-7821-ef84-96d0-d2c0318e65e1
* Version Independent ID: e40c1bbc-0411-ea87-58e6-25d53db93b3f
* Content: [Mirror Drivers - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/mirror-drivers)
* Content Source: [windows-driver-docs-pr/display/mirror-drivers.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/mirror-drivers.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

EliotSeattle, 2020-07-17:
Thanks, @terriblememory.
I'll find out what the correct location is for that special driver info and update the topic accordingly. 
Eliot

lorihollasch, 2020-08-07:
@terriblememory , thanks again for flagging this for us! We've resurrected the alias. That said, please see the note added about GDI accessibility drivers no longer being recommended.

closed 2020-08-07 by lorihollasch