MicrosoftDocs/windows-driver-docs #2369: NDIS versions on Windows Server 2019

I-gor-C, 2020-07-19:
Per this page NDIS 6.82 is supported by Windows 10 build 1809 and Windows Server 2016 and newer.
However, from build 1809 the server versions are named Windows Server 2019, as far as I know. 
Maximal NDIS version supported by Windows Server 2016 should be 6.81.

Could you please clarify this?

(same question for the NDIS 6.83 doc)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2b57e13a-8833-eb35-de7a-ca721a319e01
* Version Independent ID: 252f9733-7e17-c233-87c9-ffb82cee8612
* Content: [Introduction to NDIS 6.82 - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/network/introduction-to-ndis-6-82)
* Content Source: [windows-driver-docs-pr/network/introduction-to-ndis-6-82.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/network/introduction-to-ndis-6-82.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @aviviano
* Microsoft Alias: **amvivian**

aviviano, 2020-07-21:
Thanks @I-gor-C for submitting this issue. We'll look into this and get back to you.

aviviano, 2020-07-31:
@I-gor-C, you're absolutely right that Windows Server 2019 is the minimum version that supports NDIS 6.82 and 6.83. We've updated the relevant docs. Thanks very much for letting us know.

closed 2020-07-31 by aviviano