MicrosoftDocs/windows-driver-docs #2376: Example self-signed code signing certificate example broken

DepthDeluxe, 2020-07-23:
The example command `makecert -r -pe -ss PrivateCertStore -n CN=Contoso.com(Test) -eku 1.3.6.1.5.5.7.3.3 ContosoTest.cer` no longer generates a test certificate that Visual Studio 2019 recognizes.  I am getting the following log output when running signtool at debug level

```
1>The following certificates were considered:
1>    Issued to: Contoso.com(Test)
1>
1>    Issued by: Contoso.com(Test)
1>
1>    Expires:   Sat Dec 31 16:59:59 2039
1>
1>    SHA1 hash: 5D0BD77F91B5A483354207109AF29F27F582679C
1>
1>
1>After EKU filter, 1 certs were left.
1>After expiry filter, 1 certs were left.
1>After Hash filter, 1 certs were left.
1>After Private Key filter, 0 certs were left.
1>SIGNTASK : SignTool error : No certificates were found that met all the given criteria.
```

Software:
Visual Studio 2019 16.6.4
Windows Driver Kit 10.1.19041.1

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 33b25f0f-6898-0e7a-ae15-cf609e167a77
* Version Independent ID: b0907ff5-7e18-5dfa-72d5-02c9be59691b
* Content: [Creating Test Certificates - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/creating-test-certificates)
* Content Source: [windows-driver-docs-pr/install/creating-test-certificates.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/creating-test-certificates.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

DepthDeluxe, 2020-07-23:
Alright it looks like the command properly generates a certificate, it just places it in a location that Visual Studio doesn't look for.

The command selects to save the public + private key in `PrivateCertStore` while Visual Studio is looking for certificates in the `Personal` folder.  What makes it more confusing is that the public key is also stored in the `Personal` folder so when looking in the certificate browser, it appears that the certificate is in the right place.

It looks like directly exporting to the Personal store (`makecert -r -pe -ss Personal -n CN=Contoso.com(Test) -eku 1.3.6.1.5.5.7.3.3 ContosoTest.cer`) does not work, instead creating a new separate Personal store.  Perhaps what is missing is instructions on how to export from the private store and re-import into the personal store.

tedhudek, 2020-07-29:
Thanks for the feedback.

Try using the Certmgr tool as described in https://docs.microsoft.com/en-us/windows-hardware/drivers/install/installing-test-certificates to install the certs to the trusted store. I just pushed a doc update to move that page closer to the one you referenced in this issue (https://docs.microsoft.com/en-us/windows-hardware/drivers/install/creating-test-certificates), and also added a note to the latter with this recommendation and link to the former.  Those changes should be live within a few minutes.

Hope that helps!

closed 2020-07-29 by tedhudek