MicrosoftDocs/windows-driver-docs #2386: "Content Pending" on "Time Travel Debugging - JavaScript Automation" page

nebulous999, 2020-07-31:
This page seems incomplete - "Content Pending"

To begin with, you could link to related pages, such as:

Docs / Windows / Windows Drivers / Windows Debugging Tools / Debugging Resources /  JavaScript Debugger Scripting
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/javascript-debugger-scripting

Docs / Windows / Windows Drivers / Windows Debugging Tools / WinDbg Preview / WinDbg Preview - Scripting Menu
https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-scripting-preview#javascript-scripting

The WinDbg Preview UI would be pretty intimidating for less-technical people such as low-level QA testers or customer support.  I'm not very familiar with debugger scripting yet, but I'd be interested to know whether JavaScript Debugger scripts might help simplify TTD trace recording at all, for non-SDEs?  Or are any other features planned for this use case?
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: f7a93d71-41ee-9f70-c6a9-b06924da1f38
* Version Independent ID: 8eb20fd0-a976-0a8a-4e7f-2d0397012f41
* Content: [Time Travel Debugging - JavaScript Automation - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/time-travel-debugging-javascript-automation)
* Content Source: [windows-driver-docs-pr/debugger/time-travel-debugging-javascript-automation.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/time-travel-debugging-javascript-automation.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-08-14:
@nebulous999 - Thanks for this input and the interest in using JavaScript for TTD automation. I will see if we can as you suggest, at least offer some pointers to resources.

DOMARS, 2020-09-17:
@nebulous999 - I have a draft topic in review. Hoping to publish in the next week or so.

DOMARS, 2020-09-25:
@nebulous999 - I worked with the dev team to provide some sample code snips and other tips for using JavaScript with time travel tracing. Using the debugger and looking at code execution using time travel is a pretty complex undertaking, so not sure scripting will mask too much of the underlying complexity of what is primarily a developer tool. One resource to get started with the debugger in user mode is https://docs.microsoft.com/windows-hardware/drivers/debugger/getting-started-with-windbg and for time travel in WinDbg Preview - https://docs.microsoft.com/windows-hardware/drivers/debugger/time-travel-debugging-walkthrough. Good luck with your debugging work.

closed 2020-09-25 by DOMARS