MicrosoftDocs/windows-driver-docs #2389: Microsoft Doc Accessibility Workshop for Write the Docs 2020

tedhudek, 2020-08-01:
# Microsoft Doc Accessibility Workshop for Write the Docs 2020

We're so glad you could join us today!

We will be bringing existing docs into alignment with accessibility recommendations in the Microsoft Style Guide.  By doing so, we'll set a precedent for the community, as well as Microsoft's writers.

Specifically, we're going to update occurrences of the word click, according to the following style guide recommendations:

[Describing interactions with UI](https://docs.microsoft.com/en-us/style-guide/procedures-instructions/describing-interactions-with-ui)

Here's how to proceed:

1. Look at [file list](link).
2. In the list, pick a block of files to work on.
3. Look in discussion in the issue to ensure that no one else is already working on that block. Add a comment that you are starting work on it.
4. Either work in the browser (hint: use the `t` keyboard shortcut on GitHub to get to the file picker, where you can quickly find files) or clone [windows-driver-docs](https://github.com/MicrosoftDocs/windows-driver-docs) and work locally.
5. Open one pull request to contain all your changes. In the PR, add `#xxx` in the conversation, to link to this issue.
6. Add the `Write-the-Docs-2020` label to the PR so we can triage all the fixes later.

If you have questions, don't hesitate to message Ted Hudek on the Hopin platform during Writing Day.

Thank you for your help making our docs more accessible for everyone!

nitya, 2020-08-09:
I'll be starting with Block #1

nitya, 2020-08-09:
Picking up Block #2

tedhudek, 2020-08-09:
Go Nitya!!  We really appreciate your help.

nitya, 2020-08-09:
Picking up Block #3

vgiurgiu, 2020-08-09:
Picking up Block #5

nitya, 2020-08-09:
Picking up Block #4

nitya, 2020-08-24:
Picking up Block #6

nitya, 2020-08-24:
Reviewing Block #8 
Picking up Block #9

tedhudek, 2020-08-31:
Thanks to everyone for all the help!

closed 2020-08-31 by tedhudek