MicrosoftDocs/windows-driver-docs #2390: Invalid links on page

AndryS0, 2020-08-05:
Links does not working on this page (Page not found)

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 7932a4bc-3511-c98b-0342-ad97f2232ead
* Version Independent ID: daecf8a3-6588-2e54-bca0-c357bcffeaec
* Content: [Kernel-Mode Driver Architecture Design Guide - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/)
* Content Source: [windows-driver-docs-pr/kernel/index.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/index.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

EliotSeattle, 2020-08-05:
Working on it

EliotSeattle, 2020-08-05:
Thanks for submitting this Issue, @AndryS0! 
I updated the page pretty thoroughly, adding more relevant navigation, and testing all of the links.
Appreciate the time it took to point this out.
Eliot

closed 2020-08-05 by EliotSeattle