MicrosoftDocs/windows-driver-docs #2393: Usage of flag predefined variables in AddReg

Lukasz-Kurylo, 2020-08-06:
[Enter feedback here]
It's unclear whether flags can be used as %FLG_ADDREG_TYPE_DWORD% or need to be entered in hex.
It would be more elegant to have this in inf:
* HKR,,TypesSupported,**%FLG_ADDREG_TYPE_DWORD%**,7
 
But the sample shows:
* HKR,,TypesSupported,**0x00010001**,7 

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: acdc91cd-c89e-684c-c48b-5d66d098fa9c
* Version Independent ID: 4865d5a4-df22-1c26-f993-a661882194a7
* Content: [INF AddReg Directive - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/install/inf-addreg-directive)
* Content Source: [windows-driver-docs-pr/install/inf-addreg-directive.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/install/inf-addreg-directive.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

zlockard, 2020-09-11:
They're not predefined, you'd have to define them in [Strings], but you can do that if you'd like.

tedhudek, 2020-09-12:
Thanks @Lukasz-Kurylo for the question, and to @zlockard for the answer.  I just pushed a doc update to the page in question to add clarifying text in the Example section.  That update should appear on the live site within a few minutes.

closed 2020-09-12 by tedhudek