MicrosoftDocs/windows-driver-docs #2396: Shift+Up arrow and shift+down arrow are not documented

wcheng-msft, 2020-08-07:
Documentation says You can use the UP ARROW and DOWN ARROW keys to scroll through the command history.

It fails to mention that you can use SHIFT+UP ARROW to match partially typed commands to history.
For instance, if your command history has:
bp foo!bar
g
k
dx foo
p
t
p
t
g

and now you want to "dx foo" again
you can scroll through history by hitting up 6 times to get there, -or-
you can type dx, then hit shift+up arrow, and it should match the command history "dx foo"

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c10fd75c-5c8d-4416-db21-daceec6c47a0
* Version Independent ID: 71e35e82-323f-ee41-5454-ccd1e3131a0c
* Content: [Using Debugger Commands - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/using-debugger-commands#feedback)
* Content Source: [windows-driver-docs-pr/debugger/using-debugger-commands.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/using-debugger-commands.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-08-10:
@wcheng-msft - Thanks for pointing this out. You are also welcome to provide a docs PR with suggested wording to describe the partial match command line buffer feature. 
@aluhrs13 - FYI, using the shift command, seems like a handy thing to mention.

DOMARS, 2020-11-16:
@wcheng-msft - I tested this in WinDbg Preview and I believe that the auto complete behavior is different between the two debuggers. So I won't document this behavior here.
You can read about the classic WinDbg keyboard short cuts here: https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/keyboard-shortcuts
And WinDBg Preview here:  https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/windbg-keyboard-shortcuts-preview
Thanks for reaching out to the docs team with this suggestion, and good luck with your debugging.

closed 2020-11-16 by DOMARS