MicrosoftDocs/windows-driver-docs #2399: link broken

Behrouz-m, 2020-08-09:
the following link is not correct at target page
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/_kernel/#safe-string-functions-for-unicodestring-structures

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 3fa98807-a935-7fe0-b885-6d3d87cd1729
* Version Independent ID: ff580c01-a9c8-886d-be29-cc01e0508740
* Content: [Windows Kernel-Mode Safe String Library - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/kernel/windows-kernel-mode-safe-string-library)
* Content Source: [windows-driver-docs-pr/kernel/windows-kernel-mode-safe-string-library.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/kernel/windows-kernel-mode-safe-string-library.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @tedhudek
* Microsoft Alias: **tedhudek**

RAJU2529, 2020-08-09:
@Behrouz-m , Could you tell me, which are the links are broken, those are mentioned in this article?

Behrouz-m, 2020-08-10:
this book mark "#safe-string-functions-for-unicodestring-structures" does not go to the target

RAJU2529, 2020-08-10:
@Behrouz-m .  below link is ok for you ?
https://docs.microsoft.com/en-us/windows-hardware/drivers/ddi/ntstrsafe/

EliotSeattle, 2020-08-10:
Working on this

EliotSeattle, 2020-08-10:
Thanks, @Behrouz-m, for finding this. 
Fixed in source.
Link is now  [Safe String Functions for UNICODE_STRING Structures](https://docs.microsoft.com/windows-hardware/drivers/ddi/_kernel/#safe-string-functions-for-unicode_string-structures)
Should be live shortly.

closed 2020-08-10 by EliotSeattle

Behrouz-m, 2020-08-11:
thank you to for fixing this.