MicrosoftDocs/windows-driver-docs #2429: BulkRead does not return correct data

NateVolt, 2020-08-10:
`buffer.ToString()` does not actually return a string representation of what is in the buffer. I get `System.__ComObject`. Also, to get the returned data only, you must strip away the Bulk-In header, and the trailing alignment bytes.

``` csharp
private async void BulkRead()
    {
        UInt32 bytesRead = 0;

        UsbBulkInPipe readPipe = usbDevice.DefaultInterface.BulkInPipes[0];
        readPipe.ReadOptions |= UsbReadOptions.IgnoreShortPacket;

        var stream = readPipe.InputStream;
        DataReader reader = new DataReader(stream);

        try
        {
            bytesRead = await reader.LoadAsync(readPipe.EndpointDescriptor.MaxPacketSize);
        }
        catch (Exception exception)
        {
            ShowStatus(exception.Message.ToString());
        }
        finally
        {
            ShowStatus("Number of bytes: " + bytesRead);

            IBuffer buffer = reader.ReadBuffer(bytesRead);

            ShowData (buffer.ToString());

        }
    }
```
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: c11f0477-438f-5b76-784a-ea4284d1c756
* Version Independent ID: 8ddc42c9-a9f8-945b-8e3a-c26d6ea78496
* Content: [How to send a USB bulk transfer request (UWP app) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/how-to-send-a-usb-bulk-transfer--uwp-app-)
* Content Source: [windows-driver-docs-pr/usbcon/how-to-send-a-usb-bulk-transfer--uwp-app-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/how-to-send-a-usb-bulk-transfer--uwp-app-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

tedhudek, 2021-02-24:
Thanks again @NateVolt this is great feedback.  Since I don't know the code in question, I'm hesitant to jump in with a fix here. Can you tell me here what specific changes you're recommending - or better yet close this out in favor of a PR? Seems like a great point to get into the docs.

tedhudek, 2021-03-11:
Hi @NateVolt,

Thanks for the feedback and apologies it's taken us so long.  Your question is now tracked in a Microsoft-internal bug.  I do not have an ETA for when it will be resolved.  Because we have the bug tracking it now, I am going to close out this issue.

If you would like me to notify you if and when I get an answer, please just reply to me here and I will see it and make a note.  And you can always create a PR if you like - start by clicking Edit on the docs page.

Thanks for caring about the docs!

closed 2021-03-11 by tedhudek

mhopkins-msft, 2021-04-22:
Hi @NateVolt,

Thank you for the feedback. I've updated the code snippet. The article should be republished later today.

Thanks again!
 ~ *Mark*