MicrosoftDocs/windows-driver-docs #2434: WDDM graphics driver causing RDP mouse jitter

jantonacci, 2020-08-11:
Any plans to address the WDDM graphics driver causing RDP mouse jitter?  See https://answers.microsoft.com/en-us/windows/forum/all/windows-10-1903-remote-desktop-cursor-stutters/dc1f77db-0f35-4954-8084-937edc192558 for details.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 2e05c0fc-2bb3-964a-01e8-0e9b2f7557de
* Version Independent ID: 65380793-9c19-edbb-3879-f3dcbb8e15d4
* Content: [Migrating to the Windows Display Driver Model (WDDM) - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/display/migrating-to-the-windows-vista-and-later-display-driver-model)
* Content Source: [windows-driver-docs-pr/display/migrating-to-the-windows-vista-and-later-display-driver-model.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/display/migrating-to-the-windows-vista-and-later-display-driver-model.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @lorihollasch
* Microsoft Alias: **loriwhip**

lorihollasch, 2020-10-23:
@jantonacci , thanks for your feedback - I've passed it along to the development team. The best way to report issues is through the Feedback Hub, which I encourage you to do. Closing this issue since it isn't doc-related, thanks.

closed 2020-10-23 by lorihollasch

lorihollasch, 2020-10-23:
Just heard back from the dev team, @jantonacci. Turns out that they were aware and issued a servicing fix in Win 10 v2004. (They are not going to fix in 1903.) I also just realized that your issue was filed in August, so perhaps you're already aware of this. Apologies for my slow response!

jantonacci, 2020-11-19:
Hey great, ty. I am on Win10 build 1909 now.  I will roll back the workarounds and report back here.

jantonacci, 2020-11-19:
RDP is DOA, will have to travel to the remote site and revert RDP session changes via gpedit.msc as remote shutdown (restart) failed.  Sadness abounds.

lorihollasch, 2020-11-19:
Good luck, @jantonacci - hope you get resolution on your issues soon.