MicrosoftDocs/windows-driver-docs #2435: Dead link to docs in multiple filesystem minifilter driver samples readme files

kyanha, 2020-08-12:
https://docs.microsoft.com/windows-hardware/drivers/ifs/file-system-minifilter-drivers is a dead link, but it appears in the README.md for all of the https://github.com/Microsoft/windows-driver-samples/ filesys/minifilter/* samples.

What is the best link for someone to be directed to who has already acquired the driver sample code, but doesn't already have links to the documentation?  Can that link be made stable (i.e., redirected to the new location in the event it needs to be moved)?

EliotSeattle, 2020-08-12:
Thanks for pointing this out, @kyanha 
We recently changed our Table of Contents structure and it broke some links. I thought we got all the breaking changes, but this batch in the samples eluded us. 
The links would normally resolve to **File system minifilter drivers** 2 topics above [Filter Manager Concepts](https://docs.microsoft.com/en-us/windows-hardware/drivers/ifs/filter-manager-concepts). All of the content is still there.
We'll get to changing the link to the correct destination as our time allows.
Again, thanks!
Eliot

lorihollasch, 2020-08-17:
@kyanha , d'oh! I did indeed miss this break caused by a TOC change. I added the necessary redirect to fix it. Thanks for reporting this!

closed 2020-08-17 by lorihollasch