MicrosoftDocs/windows-driver-docs #2436: Information is out of date, MMA download is no longer available

jorticus, 2020-08-12:
This document points to the Microsoft Message Analyzer download page, but the download is no longer available and cannot be obtained.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: b034fd72-470b-76e3-e080-46cb688242db
* Version Independent ID: 3fde5448-65db-8332-2194-c3b301d0d4d5
* Content: [Capture and view USB traces with Microsoft Message Analyzer - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/capture-and-view-ing-usb-traces-with-microsoft-message-analyzer-)
* Content Source: [windows-driver-docs-pr/usbcon/capture-and-view-ing-usb-traces-with-microsoft-message-analyzer-.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/capture-and-view-ing-usb-traces-with-microsoft-message-analyzer-.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

RAJU2529, 2020-08-12:
@jorticus .  I have copy of Microsoft Message Analyzer  32bit 64bit .msi  version 4.0.8112.0

Do you need above version?

jorticus, 2020-08-13:
@RAJU2529 Thanks, but I found a copy of the MSI that I had previously downloaded.

However that doesn't help others who may try to follow this documentation and don't have a copy, and I'm concerned that since there's no longer an official copy of the tool that people will look to less trustworthy sources...

billlattimer, 2020-08-13:
As noted on the page MMA was deprecated November 2019. The purpose of leaving the document in place was to support development teams in an active dev cycle that may have been using the tool. The article will be retired today.

closed 2020-08-13 by billlattimer