MicrosoftDocs/windows-driver-docs #2437: Wrong URL used for 'Bluetooth Profile Driver functions' and 'Bluetooth Low Energy functions'

angela-yan, 2020-08-12:
[Enter feedback here]
The link for 'Bluetooth Profile Driver functions' and 'Bluetooth Low Energy functions' in the second last paragraph are both pointing to generic WDK documentation.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 8496691a-3bd6-503e-014f-94426d94d3e8
* Version Independent ID: 37997d09-0746-e388-d4a5-418d5ac34ef4
* Content: [Bluetooth Low Energy Overview - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/bluetooth/bluetooth-low-energy-overview)
* Content Source: [windows-driver-docs-pr/bluetooth/bluetooth-low-energy-overview.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/bluetooth/bluetooth-low-energy-overview.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

RAJU2529, 2020-08-12:
@billattimer,  i conformed both links are pointing to the same page,
Moreover, i cannot able to find exact replacement link.

billlattimer, 2021-02-19:
@angela-yan , @RAJU2529 thanks for catching and verifying these link errors. I'm rebuilding the pages and redirects now and it will be corrected shortly.

closed 2021-02-19 by billlattimer