MicrosoftDocs/windows-driver-docs #2444: State Value Doesn't Include Cancelled

Ben-Carpenter, 2020-08-18:
The state of the current step does not include cancelled as one of its possible values. When a label is cancelled it is labeled/set as failed. This is confusing for the contributor, they don't know if there was an actual failure or if it was cancelled intentionally. Could we please include cancelled as a value so the state is clear to the viewer.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 88bc8605-acf7-615b-3821-2f509d2a1e63
* Version Independent ID: 626ad255-cdba-6eb3-9f35-f3f688e343b4
* Content: [Get shipping label data - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/dashboard/get-shipping-labels#shippinglabel-resource)
* Content Source: [windows-driver-docs-pr/dashboard/get-shipping-labels.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/dashboard/get-shipping-labels.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @EliotSeattle
* Microsoft Alias: **eliotgra**

EliotSeattle, 2020-08-18:
Thanks for the request, @Ben-Carpenter 
This is being taken up with the Dashboard API team. Not a doc issue unless the product change is implemented, so I am going to close this.
Eliot

closed 2020-08-18 by EliotSeattle

Ben-Carpenter, 2020-08-18:
Thank you, much appreciated.