MicrosoftDocs/windows-driver-docs #2449: UsbNcm driver isn't documented

benedekkupper, 2020-08-23:
The USB class 02h, subclass 0Dh (CDC NCM) is also supported by Microsoft through UsbNcm.sys (it has been for a while on Windows 10 at least), yet it isn't documented here at all.

What's more, even though it is a USB standardized subclass, and the driver is readily available, it isn't assigned automatically to a compatible function, unlike e.g. CDC ACM. I know that this isn't the place to report that, but I haven't been able to figure out where to report such an issue so far.

---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 31d404be-d32a-fa94-3670-3586230b730c
* Version Independent ID: 5d7e6fef-e833-46c3-07f0-fc4187d7c48a
* Content: [USB device class drivers included in Windows - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/supported-usb-classes)
* Content Source: [windows-driver-docs-pr/usbcon/supported-usb-classes.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/usbcon/supported-usb-classes.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @billlattimer
* Microsoft Alias: **billla**

tedhudek, 2021-03-11:
Hi @benedekkupper,

Thanks for the feedback and apologies it's taken us so long.  Your question is now tracked in a Microsoft-internal bug.  I do not have an ETA for when it will be resolved.  Because we have the bug tracking it now, I am going to close out this issue.

If you would like me to notify you if and when I get an answer, please just reply to me here and I will see it and make a note.

Thanks for caring about the docs!

closed 2021-03-11 by tedhudek

benedekkupper, 2021-03-14:
Hi @tedhudek thanks for finally picking up this issue! Please do let me know when you have an answer for me, as it's kind of odd that Windows is currently the only major desktop OS that doesn't support plug-and-play operation for USB NCM, the fastest and most efficient standardized USB tethering protocol as of today.

tedhudek, 2021-03-19:
Hi @benedekkupper - you will be pleased to know that we updated the docs per your feedback.  See https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/supported-usb-classes, specifically the new row for `UsbNcm.sys` and the updated Miscellaneous row with the USB NCM note.

Let me know if you still have questions.

benedekkupper, 2021-03-20:
Hi @tedhudek
Big thumbs up for this note at RNDIS: "Note Microsoft recommends that hardware vendors build USB NCM compatible devices instead. USB NCM is a public USB-IF protocol that offers better throughput performance."

I have just one more question, what does "Windows Insider Preview" mean on this page exactly?

tedhudek, 2021-03-23:
Hi @benedekkupper this is the public WIP program described at https://insider.windows.com/.  Also see the Insider blog for more info -- https://blogs.windows.com/windows-insider/

Hope this helps :)