MicrosoftDocs/windows-driver-docs #2453: Link to the compress tool is dead

rossmpersonal, 2020-08-24:
[Enter feedback here]

Link to the compress tool is dead.
---
#### Document Details

⚠ *Do not edit this section. It is required for docs.microsoft.com ➟ GitHub issue linking.*

* ID: 539166bc-cb8c-eb1a-da90-5679aa7aeb03
* Version Independent ID: f66ec639-272b-7de1-0af0-e5e0bf4ce2c3
* Content: [SymStore Compressed Files - Windows drivers](https://docs.microsoft.com/en-us/windows-hardware/drivers/debugger/symstore-compressed-files)
* Content Source: [windows-driver-docs-pr/debugger/symstore-compressed-files.md](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/debugger/symstore-compressed-files.md)
* Product: **windows-hardware**
* Technology: **windows-devices**
* GitHub Login: @DOMARS
* Microsoft Alias: **domars**

DOMARS, 2020-08-24:
@rossmpersonal - Thanks for pointing out that the referenced download is not currently available. I'm investigating.

DOMARS, 2020-08-27:
@rossmpersonal - As it turns out, the compress utility has not been available for use for a number of years, so I removed references to it in the debugger docs. Thanks for filing this PR, so that we can are able to update the docs, and save other users from trying to find compress.exe in the Windows Server 2003 resource kit.

closed 2020-08-27 by DOMARS